zephyrproject-rtos/zscilib #22: Least-Squares Fitting API(s)

microbuilder, 2021-08-05:
**Least-squares fitting** is useful when calibrating certain types of sensors, such as magnetometers to fit an ellipse to a sphere. Support for **linear** and **non-linear** least-squares fitting should be added.

Possible approaches can be seen in GSL:

- Non-linear LSF: https://www.gnu.org/software/gsl/doc/html/nls.html
- Linear LSF: https://www.gnu.org/software/gsl/doc/html/lls.html

This paper may also be of use for consultation: http://www.physics.smu.edu/~scalise/SMUpreprints/SMU-HEP-10-14.pdf

microbuilder, 2021-08-24:
Closing for now since non-linear LS and min volume enclosing have too many missing dependencies to implement at this point, but linear least squares and correction functions to adjust magnetometer vector arrays have been implemented.

closed 2021-08-24 by microbuilder

jdwang001, 2024-03-28:
Hello, I'm currently porting an algorithm library to an MCU , which is implemented based on GSL (GNU Scientific Library). 
The library includes a function 
```
void polyfit(const double *xvals, const double *yvals, int n, int order, double *coeffs),
```
commonly found in NumPy and MATLAB. I'm not very familiar with ZSCLib and GSL. 
I would like to know how to proceed step by step to use polyfit in ZSCLib. 
Could you provide me with a rough idea or guidance?