zephyrproject-rtos/zscilib #29: Follow up on the zscilib blog post

fgervais, 2022-03-31:
Hi @microbuilder,

This is not an issue and I'm basically just hijacking this communication channel so feel free to disregard.

I read your blog post on the zephyr web page and I'm sort of walking on your footsteps trying to get a microcontroller to calculate the color temperature. I actually thought this was going to be a quick project before reading your post.

I'm not up to speed yet here but I'll risk a couple questions.

I'm trying to get a feel for what kind of sensor I would need to get a good light temperature measurement and so be able to match a studio light to the sun.

You mention that an RGB sensors is not precise enough to get a good temperature calculation.

1. I'm guessing a sensor with XYZ output like the [TCS3430](https://ams.com/tcs3430) would be as bad as a sensor with RGB output right?

One of the critic you have with RGB sensors is that the response of the individual 3 channels is too wide frequency wise and also the fact that they overlap.

2. Based on your experience, would 8 channels instead of 3 like the [AS7341](https://ams.com/en/as7341) be enough to get a decent temperature reading so that our eyes would not see the difference between a light that we sensed the temperature and another light we set to the same sensed values?

3. If you were so guess the lowest cost sensor that could give a precise temperature reading what would it be?

Thank you if you get time through this and give some thoughts, if not, thank you for the blog post and the zephyr library.

microbuilder, 2022-04-02:
I've tried some of the 8-9 channel devices, and even with those I struggled to get better than +/-10-20% of a decent spectrometer for measuring CCT, often closer to +/-20-25%. It could be the code, but I spent a LOT of time and effort on this, and ultimately my conclusion was that the only way to solve this with a reasonable margin of error (<10% from a known-good result from a calibrated spectrometer) is to use a linear image sensor and design a spectrometer that separates at a minimum dozens of tight wavelength ranges into individual datapoints, calibrating the sensor before use and compensate the datapoints accordingly.

There are some commercial devices for this, though not inexpensive nor easy to order:

- https://www.hamamatsu.com/jp/en/product/photometry-systems/mini-spectrometer.html

But if you want quality results, there isn't a $5-25 sensor that will deliver that in my experience (though I'll be happy to be proven wrong). Designing your own spectrometer is possible, but involves a bit of mechanical and optical engineering as well, which is the hurdle I ran into. It's a solvable problem for one person ... but it's non-trivial, and requires a bit of dedication and financial investment.

Some books I purchase on the theoretical side that you may find useful:

- Introduction to Spectroscopy (Pavia, Donald, Lampman, Gary, Kriz, George, Vyvyan, James)
- Spectrograph Design Fundamentals (James, John)

I know I have others, but those are the ones I can find in my order history on Amazon.

Sorry of that's not the answer you want ot hear, but that's the conclusion I came to myself.

fgervais, 2022-04-02:
Nice, thank you very much for this concentrated page of experience, much appreciated. It will accelerate my progress for sure.

I didn't see the mini spectrometers in my researches this looks quite interesting, I'll have a look.

Also thank you for the book suggestions, I'll check these out as well.

I'm making a little board for the AS7341 this weekend so I'll probably give it a try. I will let you know if I ever get good results with this route.

closed 2022-04-17 by microbuilder