zephyrproject-rtos/zscilib #32: Matrices: Other Facotrisations to Consider

microbuilder, 2022-05-19:
Some other matrix factorisations to consider for implementation in the future:
https://nhigham.com/2022/05/18/the-big-six-matrix-factorizations/

sdalu, 2022-08-17:
Example of LU decomposition:
https://en.wikipedia.org/wiki/LU_decomposition#C_code_example

mkschreder, 2022-08-31:
@sdalu control systems toolbox for zephyr provides this here: https://github.com/swedishembedded/control/blob/main/src/linalg/lup.c

We also have a whole range of decompositions and advanced algorithms.. 

https://github.com/swedishembedded/control/blob/main/src/sysid/sr_ukf_parameter_estimation.c
https://github.com/swedishembedded/control/blob/main/src/linalg/svd_golub_reinsch.c